mnemonicdevices/md001 #34: Chaos-based modulation

cdromain, 2023-09-05:
## Feature request & motivation
Would be amazing to have chaos-based modulation as a wave option besides traditional waveforms and random, like a Lorenz or Chua strange attractor. Chaotic modulation can be more interesting than pure random as there is some kind of pattern going on, within finite boundaries, which makes it a mix of random and deterministic. Love those drunk/super slow modulation sources for drones and I think such thing would feel right at home within the Juniper, extending its palette quite a bit into less ventured territories, making it even more versatile.

## Details
- Ideally we should be able to control the rate from super slow (minutes) to faster and this should be free-running.
- Strange attractors output 3 CVs (x, y & z) so we could choose which one we want to output ? Or use 3 outs to have all signals available at the same time ? Or mix the signals ? Or would it be possible to use Gin's additional inputs as outputs ?

## Modules implementing such chaos-based modulation
- [Triple Sloth — Nonlinearcircuits](https://www.nonlinearcircuits.com/modules/p/triple-sloth) : *This module contains 3 Sloth chaos circuits. Each one runs at a different rate; Torpor takes approx. 15-30 seconds to travel around 2 strange attractors, Apathy takes 60-90 seconds and Inertia takes 30-40 minutes.*

- [Ornament & Crime Low-Rents](https://ornament-and-cri.me/user-manual-v1_3#anchor-low-rents) : *Low-rents is a port of the [Lorenz attractor](https://en.wikipedia.org/wiki/Lorenz_system) modulation generator from the easter egg in the Mutable Instruments Streams module, to which [Rössler attractors](https://en.wikipedia.org/wiki/R%C3%B6ssler_attractor) have been added. Two independent function generators are provided (referred to here as Generator 1 and Generator 2), with each generator calculating both the Lorenz and Rössler functions simultaneously, using the same phase accumulator, but with the rate/speed of each generator independently settable. Both the Lorenz and the Rössler functions output three values (x, y & z), and various combinations of these can be mapped to the four output channels. The chaotic strange attractors work best as slow modulation functions.*

- [Orbit 3 – Joranalogue Audio Design](https://joranalogue.com/collections/eurorack/products/orbit-3) : *In everyday usage, 'chaos' means 'disorder, randomness'. In mathematics however, chaos has a very specific meaning. Chaos theory deals with systems exhibiting highly irregular behaviour, seemingly random yet completely deterministic. Applying this to the world of modular synths opens up a whole new world of modulation opportunities, introducing a natural irregularity to otherwise purely ‘mechanical’ music, yet with much more control and repeatability than just simple randomness. In chaos theory, the combination of all possible oscillations at a specific set of parameters is known as the ‘attractor’. Orbit 3 incorporates a modified electronic analogue of the classic double-scroll attractor. Since it contains some (bi)stable regions in addition to chaotic ones, this type is known as a ‘strange’ attractor.*

- [Zlob Modular: Diode Chaos](https://zlobmodular.com/product/diode-chaos/) : *Diode Chaos is a 3hp analog chaotic low frequency voltage generator. It is a unique module because it has never existed as an electronic music instrument until now. The module is based on a circuit from a paper called “A simple chaotic circuit with a light-emitting diode” written by Volos, Wang, Jafari, and Kapitaniak. Diode Chaos produces 3 phases of interrelated morphing voltages including a chaotic trigger out. The range is 20-30 seconds a cycle up to audio rate, although only the X out and TRIG out go up to audio rate. The frequency is controlled by the RATE knob. Each output exhibits different behavior at different rate settings. X,Y, and Z outs like to stay in the 4vpp range although at different RATE settings the outs can reach 10vpp. The TRIG out chaotically releases 0-10v triggers although sometimes it will spit out smaller voltage peaks.*

- [Mutable Instruments Stages Qiemem Alt. Firmware](https://github.com/qiemem/eurorack/tree/bipolar/stages) : *Double-scroll attractor, a smooth chaotic system similar to the Lorenz system (single, looping, non-gated)*

## Additional references
- [Attractor - Wikipedia](https://en.wikipedia.org/wiki/Attractor)
- [Lorenz system - Wikipedia](https://en.wikipedia.org/wiki/Lorenz%20system)
- [An introduction to Chaos Generators by Chris Meyer](https://www.elby-designs.com/webtek/panther/if108-chaquo/chaos-generators/chaos-generators.htm)
- [FAQs — Nonlinearcircuits](https://www.nonlinearcircuits.com/faq)
- [Chaos Theory - Ian Fritz](http://ijfritz.byethost4.com/Chaos/ch_over.htm) - [Overview](http://ijfritz.byethost4.com/Chaos/ch_close.htm)
- [The HYPSTER Hyperchaos Generator, Ian Fritz, Jan 2018](http://ijfritz.byethost4.com/Chaos/fourth_worddoc_1.pdf)

ostenning, 2023-09-06:
I really love this feature request. I think you are right, it would suite Juniper quite nicely - providing I can find a way to make it fit nicely with the workflow, that will be the main challenge here, especially if multiple outputs are required as you suggested.

I'll need to do some more reading into this topic to get a greater understanding of what is required, thanks for the links they are helpful.

> Or would it be possible to use Gin's additional inputs as outputs

Unfortunately Gins inputs cannot be used as outputs

cdromain, 2023-09-27:
Sounds great Oliver !
Hope Chaos can find its way into Juni soon 😃 
Lemme know if you have any question.
Editing my post above to add Disting as another example of chaos-based source.

cdromain, 2023-10-01:
Added another chaos module to my original post, Vult Caudal - discovered it this week-end and should get my hands on one of the first units in the next days !

cdromain, 2023-10-09:
Vult Caudal arrived 3 days ago 🔥 Super interesting, a totally different take on the chaos modulation as expected ! Got one of the first 4 modules out there and am waiting for Leonardo to publish more info and documentation about the different modes.

Added another module to my original post, Neutron Sound Dust of Time.

cdromain, 2023-10-10:
Updated my original post again, with more information by Jürgen Meier (in German 😄 )

ostenning, 2023-10-10:
> Updated my original post again, with more reference information by Jürgen Meier (in German 😄 ) and 2 YouTube videos.

Thanks! I'll try to find some time to study this soon

cdromain, 2023-10-12:
Can't wait to see what this inspires you ! As you understood I'm passionate about this so don't hesitate to reach out to discuss and brainstorm together 🚀

cdromain, 2024-06-01:
@ostenning Your comment and 0.6.0 milestone got me real excited man 🤩
Patience is a virtue often rewarded - I'm looking forward to bringing some chaos to Juni's powerful modulation arsenal 🔥
As always, happy to help in any way I can and discussing it further - you know where to find me !
Chat soon, cheers

ostenning, 2024-06-04:
@cdromain I've got a concept ready.

There is now a new added `Chaos` waveform which computes the 3 Lorenz equations, X Y and Z are morphable with the Morph parameter (similar to how Sine and the other waveforms work).

The frequency changes how quickly the equations are calculated, which works similarly to how the `Random` waveform works. The higher the frequency, the faster the waveform moves. Phase dictates when it moves from one value to the next.

I think all that works fairly well - my main concern is that each channel operates independently from each other, and while you can Morph X-Y-Z, you cannot represent X-Y-Z all at the same time on 3 separate channels. 
The only way I can see something like this working is if I reconceptualize "Linked Channels" which will require architecture changes to how Juniper works.

cdromain, 2024-06-04:
@ostenning You rock man 🙌🏼 Testing the alpha as I write this !

The linking of channels would be a great next move (ideally using channel N to control the 3 signals that would then be distributed on channels N, N+1 and N+2), but it's gonna be really useful already even unlinked ! (which is the case in other single-channel implementations like Stages for example)

How are X, Y & Z distributed on the Morph parameter ? I.e. Z at -50... ?

I have one feature request already - the possibility to view the attractor(s) as a XY map instead of the usual waveform oscilloscope. That would be really cool to look at, useful to understand and follow the attractor, and would also highlight the special nature of the signal vs a classical LFO. Like what is done in the Dust of Time :

[image]

Cheers

ostenning, 2024-06-04:
> The linking of channels would be a great next move (ideally using channel N to control the 3 signals that would then be distributed on channels N, N+1 and N+2), but it's gonna be really useful [...]

I think Linking channels is not out of the question, as I already link the Recorders, but it requires a lot of re-thinking about how linked channels work in general. E.g. would there be different links?  one for recorders, one for Lorenz attractors? Or would there be just one link for everything? Linking could also be handy for pitch and chords. Obviously though, some settings wouldn't be affected, and then trying to create a UI that ties it all together elegantly will likely require a bit of work. [I'll create a ticket for it for further discussion](https://github.com/mnemonicdevices/md001/issues/75)

> How are X, Y & Z distributed on the Morph parameter ? I.e. Z at -50... ?

Currently -50 is `X`, 0 is `Y` and +50 is `Z`. They are Morphed together with the same mixing equation used for `CV` and `Out` waveforms.

> I have one feature request already - the possibility to view the attractor(s) as a XY map instead of the usual waveform oscilloscope.

Hehe - yes, I thought about this almost immediately, the time axis oscillscope doesn't really convey whats going on with the attractor, plotting X against Y, or Y against Z, would be more helpful - but this would only be available for the Lorenz attractor, so I'm not sure exactly how to reconcile the UI/UX of switching between oscilloscope and X-Y plot.

ostenning, 2024-06-05:
@cdromain So I've implemented [Copy-Paste](https://github.com/mnemonicdevices/md001/issues/37) of channels, which actually "unlocks" being able to get all 3 channels of the Lorenz equations.

The workflow is like this:
1. Configure a channel to use Lorenz waveform, set Morph to `0` so its using `Y`, also set the frequency as desired
2. Copy the channel, and paste it twice into two other channels. This will copy the exact Lorenz state over to the other channels
3. Use morph to configure the other channels to `X` and `Z`.

The downside, of course, is that you cannot modify the frequency all at the same time, so if you change one they will go out of "phase" and you'll need to re copy-paste them.

I'm not sure this is the best workflow, but its a work-around at least for the moment.

cdromain, 2024-06-05:
@ostenning Thanks man, that's great news - glad to now be able to copy/paste channels, great workflow improvement 🙌🏼
And a clever workaround indeed !

ostenning, 2024-07-09:
@cdromain I've got a pretty rudimentary X-Y plot as you suggested, it works pretty well. Not finished yet, but it will be in 0.6.0 👍

cdromain, 2024-07-10:
Hey @ostenning thanks, that's exciting news ! Can't wait to test the new FW 🙌🏼

closed 2024-08-05 by ost-ing