mnemonicdevices/md001 #37: Copy and paste settings between channels

cdromain, 2023-09-27:
## Feature request & motivation
Instead of having to mult a Juniper signal to have several copies of it, would be great to be able to easily copy and paste channel settings from one channel to the other - both to have the same signal on several channels/outputs, but also as a quick way to use one channel settings as a starting point for a variation of it on another channel.

## Implementation
Will need to think about how this could be implemented interface-wise but using the channel menu could be simple enough.

ostenning, 2023-09-30:
Copy-Paste is a great idea, and will work quite easily with the 0.4.0 changes.

cdromain, 2023-10-01:
Great, thanks Oliver - can't wait for this simple feature that will prove to be super useful I'm sure !