mnemonicdevices/md001 #38: Use Gin knob as an alternative to edit Juniper values

cdromain, 2023-09-27:
## Feature request & motivation
Would love to be able to use Gin knob as an alternative/complementary gesture to modify values when editing on Juniper, for a different, more performance feel when changing values.

ostenning, 2023-10-10:
Hi @cdromain on initial thought I think this will create a UX clash, e.g. if you assign a modulation source to the POT and then you are modifying values with the POT, there could be some overlap that needs additional cognitive load to deal with. I want to try to keep things as straight forward as possible, and due to the freely assignable nature of inputs I fear that it could be overwhelming and/or confusing for some people

cdromain, 2023-10-12:
Hey Oliver, I totally get this potential UX conflict issue and am totally with you on keeping things straight forward.
Need to work more with the POT as it stands and think more about it.

ostenning, 2023-10-18:
I'll leave this for now then and maybe we can revisit it in the future

closed 2023-10-18 by ostenning

cdromain, 2023-10-19:
Yes I'll think about it more and see if I can figure something out that would keep the best of both worlds !