mnemonicdevices/md001 #39: Random walk alternative mode for random waveform

cdromain, 2023-09-27:
## Feature request & motivation
I love the random walk signal created by the [Sonic Potions Mal-2](https://sonic-potions.com/mal-2) module. This creates super smooth modulation perfect for drones, as the idea is to have a very small random change at every step never far from the last value - basically tossing a coin for a unit change (positive or negative with equal probability) at every step, so always moving in a random fashion but just a tiny bit at a time.
Tried to approximate it using Juniper random waveform, with max slew, slow mod and low amplitude, but unless I'm mistaken I can't reproduce a random walk function.

## Implementation
This alternative random mode could be nicely enabled with a long press on either WAV or MOR when using the random waveform. Alternatively this could be a random option setting in the channel menu.

## References
- [Random walk - Wikipedia](https://en.wikipedia.org/wiki/Random_walk)

cdromain, 2023-09-28:
Edited my post after more research and thought about the idea 🚀

ostenning, 2023-10-18:
@cdromain  Thanks once again for the nice write up! 

For context the current random waveform generation is periodic, after some time period as determined by the frequency, a new random 16bit number is calculated, then moving from the old value to the new value (the gradient) is determined by the `Morph` parameter. A low morph will result in a more stair-stepped pattern, while a larger `Morph` results in a smooth transition between the values.

I haven't dug too deeply into your references yet, but if I understand correctly, the major difference here is how the 16bit number is calculated, currently it is entirely random, while your suggestion sounds like there would be some smaller scoping of the randomly generated number. Right?

It sounds like we could add another Morph component for this, similar to the Envelope waveform.

I'll assign this for the next release :)

cdromain, 2023-10-19:
My pleasure 🙂
Yes, basically at every step, instead of the 16bit random generation, toss a coin and add +x or -x to the current value, with x equal to a small value, that could be chosen by the morph parameter ? Can't wait for v0.5 ! 🔥

ostenning, 2023-10-20:
@cdromain The step length would still be determined by the frequency but I think that's OK, if a user wants super fine grained movements, they can just turn the frequency up high

cdromain, 2024-06-06:
Hey Oliver, it's great to see Juni getting more love lately, so a lil' bump to dig out this old feature request of mine as I've been thinking a lot about this type of signals lately.

I think a great inspiration here is what the [VCV Rack Random](https://vcvrack.com/Free#Random) module offers, i.e. the possibility to have a blend of totally random and the previous value :
> _"When the internal random source generates a new value, it mixes the previous value with a random value, with an amount specified by the RND slider. If RND is 100%, the result is fully random and not dependent on the previous value. If RND is 0%, the result always equals the previous value. If RND is between these extremes, the result is random but near the previous value by some margin."_

This goes in the Random walk direction - i.e. having the signal changing randomly just a bit at each step, for a continously randomly evolving signal, always different but related, not far from the previous value - but with even more nuances in how much we want the signal to change. Having such a blend modulated could open really interesting avenues.

If this becomes a distinct waveform type, ideally we'd like to be able to control the previous/random blend while keeping the control on how stepped/smooth the output is - so maybe blending between random and previous between -50 and 0 (stepped) and then between previous and random but smooth between 0 and +50 ? Just brainstorming here and it's getting late 😄 
Another option could be to add this to the current Random mode and alternate between stepped/smooth and previous/random control with a long MOR button press.

Hopefully this could be quite quick to prototype ? That would be super useful pretty please 🙌🏼

ostenning, 2024-07-10:
I'll add this in the 0.6.0-alpha-5 release

cdromain, 2024-07-10:
Wicked, can't wait mate ! Patience is always rewarded, right 😃 Cheers

ostenning, 2024-07-11:
@cdromain 

> I think a great inspiration here is what the [VCV Rack Random](https://vcvrack.com/Free#Random) module offers, i.e. the possibility to have a blend of totally random and the previous value :

> Another option could be to add this to the current Random mode and alternate between stepped/smooth and previous/random control with a long MOR button press.

Your original suggestion of the pure incremental drunken walk was going to be given its dedicated waveform type "Drunken", but your updated suggestion of having the ability to blend between the previous value and a random value tends me to think that adding an additional MOR parameter for the current Random waveform makes the most sense

cdromain, 2024-07-11:
I have to say I love the idea of a "Drunken" waveform type 😄 But yeah the VCV Random-inspired blend opens up more possibilities and should yield a pretty good random walk approximation ! CAN'T freakin WAIT 🔥🙌🏼