mnemonicdevices/md001 #40: Higher max (audio rate) frequency

cdromain, 2023-10-01:
## Feature request & motivation
The possibility to have unsynced free-running modulators is a great addition in 0.4 that I had been asking for. Thanks ! Works great for audio rate modulation, which made me wish max frequency could go higher than the current 128Hz ceiling. Is it something that could be done ? Going to 1KHz maybe ?

And yes I tried listening to the LFOs running at audio rate - found the random waveform especially inspiring :)

ostenning, 2023-10-10:
There are some philosophical and technical limitations. 

While totally possible to change the upper limit of the frequency to 1kHz, because of the sample rate, the audio will alias heavily and wont provide in my opinion, good enough results (especially the Sine waveform). 

Generally I would prefer a more concise honed feature set specific to CV and I think Juniper does a lot already.

cdromain, 2023-10-12:
I understand and totally respect that you want to keep the focus on CV modulation versus audio oscillations Oliver, but my feature request was actually CV focused (audio rate modulation).
Maybe then 1K was too much too ask given aliasing considerations, but could we increase the max frequency rate a bit more ? That should prove to be useful for exploratory audio rate modulation purposes. My first tests in that direction were pretty interesting.

closed 2025-01-15 by ost-ing