mnemonicdevices/md001 #65: Stepped and "spring" like waveforms

CV-Gate, 2024-02-04:
It would be awesome to have stepped waves (frequency and nr. of steps) like the ones in NI Clep Diaz and other more "springy" or "bursty" like the ones in TINRS Wobbler 2.

ostenning, 2024-02-18:
Hey, thanks for the suggestion - I really like this idea, and it should be fairly straight forward to implement

ostenning, 2024-02-27:
Thanks @CV-Gate, done and added to 0.5.0 ✅

closed 2024-02-27 by ostenning

hughc-hub, 2024-08-17:
hey!
I have a suggestion regarding this type of wave, or to see how it could be done: a step wave where I can define the step value in volts so that it can be +1V, 0V, or -1V... or another value from a list that I provide (essentially it would be like giving quantized values)... and that this wave changes from one value to another with a long time value that I can define, including that after a certain number of measures or time, it repeats... 
Is this possible?

ost-ing, 2024-08-28:
@hughc-hub sorry for the late reply. Its an interesting idea, how would the steps be defined? I guess it could be done on computer and stored on the sdcard fairly easily. I think we should make a new issue for this idea to discuss further

hughc-hub, 2024-08-28:
I could imagine it as sort of a precision adder (or negative values also), where the waveform would allow us to raise or lower octaves, for example,( in a Morphagene you can play at 100% speed, half, 1/4, ... or reverse playback using similar in negative).
All these changes in it are made in multiples of V/oct. So ideally, it should be possible to define these values in the menu itself. A kind of "scale" where we specify which "values" it can use, and then the waveform will apply one or another based on a certain probability. It would be like note quantization, but where the "notes" are octave multiples.