mnemonicdevices/md001 #66: Allow applying modulations to recorded waves

CBluesF, 2024-02-18:
**Is your feature request related to a problem? Please describe.**
Once a recording has been made it is fixed in place and e.g. the amplitude cannot subsequently be changed, wavefolding cannot be applied, etc.

**Describe the solution you'd like**
Allow functions to be applied to an already recorded wave, as if the recording was an input.

**Potential ramifications or implications**
If this was on by default, recording a wave while modulations are running (to "save" those modulations) would then apply them _again_ once the recording is done. I would suggest having this as a toggleable option, off by default, to avoid that. Alternatively, the modulations active for the unrecorded wave could save and a fresh set open up for editing while the recording is playing. 
 
**Additional context**
Add any other context or screenshots about the feature request here.

cardinal-bin, 2024-06-30:
There's a potential interesting option here for what happens when pressing record again - a nice option would be to allow the modulated recording to be recorded itself, overdubbing the existing recording (and again, resetting the modulations to avoid doubling up). 

So we could go from: modulated wave => fixed recording of modulated wave => modulated recording => fixed recording of previous modulated recording.

ostenning, 2024-07-09:
I thought about this a while ago when I first implemented the recorder and decided against it to try to maintain simplicity. With the `out` channel this functionality can be achieved (although it will use 2 channels rather than just 1), so I'm leaning towards leaving it how it is, at least for the time being.

cardinal-bin, 2024-07-13:
@ostenning  Perhaps instead of allowing all modulations, just some manual changes could be allowed? In particular it would be very useful to be able to attenuate or amplify post-recording.