mnemonicdevices/md001 #67: Dokumentation

woowooly, 2024-03-21:
**Describe the topic for investigation**
A clear and concise description of what the topic is.
Is it possible to bring up your Dokumentation to a level what evreybody can understand whats goin on?
Thanks

cdromain, 2024-07-10:
Hey guys, discussed about it with @ostenning at Superbooth, I'd be glad to help flesh out & expand on the existing documentation to make it more intuitive and user-friendly. More examples, tips & tricks and suggestions of things to try would be nice as well ! Cheers

ostenning, 2024-07-11:
I will make the docs open-source for more user contributions, patch ideas etc

woowooly, 2024-07-11:
Hi Oliver... Thank you.
Please bring the Quantizer up to date. I need it to pass, maybe only Note C E and B ... ...
Your work ist great and very cool.
Bg Lars

Von Outlook<http://aka.ms/weboutlook> gesendet.

ost-ing, 2024-07-22:
> ...Please bring the Quantizer up to date

The quantizer is being improved, it will be in the next release and will be out in the next week or so, you can pick a custom scale for each channel 👍

ost-ing, 2024-07-24:
Documentation is now open-source and available at https://github.com/mnemonicdevices/docs . For any issues regarding documentation please open them there. Thanks

closed 2024-07-24 by ost-ing