mnemonicdevices/md001 #68: Offset doesn't affect the unipolar waveforms

CV-Gate, 2024-03-24:
When I set a channel as "unipolar" and I modify "offset" the percentage on the screen changes, but the waveform range is not affected.

**To Reproduce**
Set a channel as "unipolar," choose a sin waveform, and manipulate the offset. Despite the percentage changes, the waveform stays in the same place.

**Expected behavior**
The affected waveform's offset must change when the offset is manipulated, even if it's unipolar. The waveform is expected to be compressed depending on the amp settings.

**Versioning (please complete the following information):**
V 0.5.0

ostenning, 2024-03-26:
Thanks @CV-Gate I've just tested this and reproduced the bug. I'll have it fixed soon.

ostenning, 2024-03-28:
Fixed in 0.5.2

closed 2024-03-28 by ostenning