mnemonicdevices/md001 #69: Select + channel buttons combo time setting

cdromain, 2024-04-24:
## Feature request & motivation
When using the Select + channel buttons combo to select a channel, I way too often end up switching parameters when I actually want to switch channels (e.g. T&H instead of channel 1). What was a detail has become a bit irritating, given how much I use my beloved Juni 😃 (I know we can hold the Select button but that's not comfortable in my setup.)

## Implementation
A setting in the global settings would allow the user to modify (increase) the time available to push the channel button after the Select button. Seems like the default time we have to get the desired behavior is around 2 seconds. I think 3 to 5 would work better for my preferences.

ostenning, 2024-05-13:
Thanks for the feedback as usual! Yes, I will improve this for the next release

CBluesF, 2024-05-27:
Just chiming in to add that it would be great to be able to reduce that time, too. I have the opposite problem, I usually use hold select + channel, and then when I try to choose a parameter change channel by mistake!

ostenning, 2024-05-28:
@CBluesF noted and thanks for the feedback - I've been working on this the last few days, it'll be in the next release

closed 2024-08-05 by ost-ing