mnemonicdevices/md001 #70: The current method of switching between LFO and function generator mode seems wrong

MvKeinen, 2024-05-09:
**Describe the topic for investigation**
... unless I miss something.

Right now we are switching between LFO and function generator mode by defining a trigger source in the channel menu. AFAICS There is no way to restart a continuously running LFO. This is very important for many applications.

Solution for the channel menu:
- Trigger source always starts/restarts the LFO when set.
- new toggle parameter (only appears when trigger source is set): "one-shot mode" ON/OFF

Now only when the new parameter (one-shot) is set to ON the channel switches to function generator mode.

As I find this to be a basic feature in LFOs I wonder if I missed something.
If there is need I'd be happy to describe possible applications of restarting a continuously running LFO in more detail.

Thanks for this amazing module!

ostenning, 2024-05-12:
Thanks for the suggestion - I will add support for this in the next release

ostenning, 2024-05-29:
@MvKeinen 

I'm trying to figure out the best UX for this. I'm considering cleaning up the Channel Settings Menu and moving some of these configuration items into a more appropriate spot:

`Trigger Mode` would now be in `Phase Menu`
`One-shot Mode` would be in the `Phase Menu`
`Synchronized/Unsynchronized` would be in the `Frequency Menu`.

This way the settings are better located.

What do you think of this idea?

ostenning, 2024-05-29:
Another caveat - using an external trigger to restart the LFO will only work when the LFO is not phase-locked (set to "Free"). This is because the Phase locked mode by definition forces the phase to be in alignment with the other channels.

MvKeinen, 2024-05-29:
@ostenning 

> @MvKeinen
> 
> I'm trying to figure out the best UX for this. I'm considering cleaning up the `Channel Settings` Menu and moving some of these configuration items into a more appropriate spot:
> 
> `Trigger Mode` would now be in `Phase Menu`
> `One-shot Mode` would be in the `Phase Menu` 
> `Synchronized/Unsynchronized` would be in the `Frequency Menu`. 
> `Unipolar/Bipolar` would be in the either in `Amplitude Menu` or `Offset Menu` TBD

I think both solutions have advantages. Having all settings in one menu might be faster in cases where you want to setup multiple channels at once and you have a clear plan of what you want to do with say 4 channels. On the other hand, having the configuration items inside their fitting parameter menu would definitely help when experimenting and trying out new things, as in getting deep into the functionality of a parameter.

> 
> This way the settings are better located.
> 
> What do you think of this idea?

Is both possible?
Having all those parameters in the settings menu AND some of them in their appropriate parameter menu.

MvKeinen, 2024-05-29:
> Another caveat - using an external trigger to restart the LFO will only work when the LFO is not phase-locked (set to "Free"). This is because the Phase locked mode by definition forces the phase to be in alignment with the other channels.

The problem with "phase" is that it is implemented in so many different ways throughout synthesis that it's hard to keep track and know what's going on. The Juni handles phase differently as I'm used to and I haven't completely wrapped my head around it yet. I think the main difference is that the Juni has a global phase derived from the clock/bpm and every channel not set to "free" and not being unsynchronized follows that phase. 

I would more expect every channel to have its individual phase derived from the "frequency" parameter (+modulation) of the channel. 
The "phase" parameter would set the start phase inside one cycle. AFAICS the "free" setting isn't necessary then. That way any setup can be triggered externally.

I think that's a very important topic with the juni. Global phase locking is certainly very powerful, but when you modulate extensively it can be a hassle.

I didn't have the time I wanted with the Juni in the last few months, so I'm still not clear about a few things. I have much more time now and will think about these things extensively.

What I think is important:
- Having a LFO with smooth (not stepped) frequency modulation (already possible)
- resetting that LFO at a certain phase of its own cycle (eG sine at 90° start phase starts at maximum) with an external trigger.

When an LFO is frequency modulated a fixed phase lock makes little to no sense anyway.

ostenning, 2024-05-29:
Can you please send me an email, I have an alpha test firmware you can try out with this feature

ostenning, 2024-05-29:
> The problem with "phase" is that it is implemented in so many different ways throughout synthesis that it's hard to keep track and know what's going on. The Juni handles phase differently as I'm [...]

Currently the phase is mainly derived by the "runtime" of the device as well as the frequency of the channel. Changing this, I think deserves its own ticket really - as it would fundamentally change the way Juniper implements phase.[ I've created this ticket](https://github.com/mnemonicdevices/md001/issues/74) where we can discuss further

MvKeinen, 2024-05-29:
hey Oliver!

My Email is ***@***.***

I'd love to try an alternative Firmware

cheers David

ostenning, 2024-05-30:
@MvKeinen I don't think you can pass email on here. Please send me an email from the mnemonicdevices.io support page

cardinal-bin, 2024-06-30:
>What I think is important:

>Having a LFO with smooth (not stepped) frequency modulation (already possible)
>resetting that LFO at a certain phase of its own cycle (eG sine at 90° start phase starts at maximum) with an external trigger.

I would add that it would be preferable to maintain the existing functionality as an option after incorporating this. Personally I use the present implementation to line up LFOs with certain parts of bars/phrases, it's very useful. 

One possible option would be to have a received trigger actually _set_ the phase 'start point' to the time when a trigger is received. This would need to be sufficiently un-quantized that it sounds smooth, but would allow the existing phase implementation to persist while also allowing sync triggers.

closed 2024-08-05 by ost-ing

hughc-hub, 2024-08-17:
In relation to this functionality (which I think is one of the most interesting), I was thinking of implementing the idea somewhat derived from other modules like Zadar or Pamela Pro. In addition to what other colleagues have mentioned, I see 3 possible use cases.

First, that via a CV in, when there is a gate, the wave of that channel restarts. Another option is to implement two duration options for the loop within the channel (similar to how it's done in Zadar or Pamela Pro).

The other two options: You can define the time in seconds/minutes that the wave will take to complete and restart. Or define how many measures (using the clock) you want it to be... Both modes are very useful for generative music and modulations that are aligned with a melody