mnemonicdevices/md001 #72: Amplitude units to become Voltage

ostenning, 2024-05-28:
The 16bit DAC allows for very fine grained voltage amplitude control, but the current way (percentage) of changing amplitude is quite crude. Additionally the fine/grain control for amplitude seems overlooked.

- Amplitude would be in terms of voltage
- Voltage would be peak-to-peak and displayed as "Vpp", e.g. "10Vpp" would indicate -5V to +5V. 
- A Unipolar signal would be 5Vpp (0V to 5V). 
- Inverted waveforms would have a Voltage of "-Vpp".
- Course control would be 100mV steps, e.g. 4.1Vpp -> 4.2Vpp (pending testing)
- Fine control would be 10mV steps, e.g. 4.01Vpp -> 4.02Vpp

MvKeinen, 2024-05-28:
That would be a great improvement! I'm looking forward to this!

MvKeinen, 2024-05-28:
Maybe it would make sense to do the same with the offset parameter.

ostenning, 2024-05-29:
@MvKeinen absolutely - I've updated this issue

ostenning, 2024-05-29:
I've opted for 1mV steps being the smallest for now - accuracy depends on calibration but will need to do more testing, I may have to revise this

CBluesF, 2024-06-13:
An option to adjust in 1V or 1/12V increments would be amazing for transposing incoming or random sequences.

closed 2024-08-05 by ost-ing