mnemonicdevices/md001 #75: Linking channels

ostenning, 2024-06-04:
**Describe the topic for investigation**
Linked channels

This is useful for Recorders, Lorenz Attractors and pitch / chords. A lot of unknowns about UI and implementation that need to be figured out.

cardinal-bin, 2024-06-27:
Reading through the discussion on the Lorenz attractors thread I have a related suggestion for how 'linking' could work. Although I'm not exactly sure how your implementation would accept this, it makes sense from a UX perspective and could be relatively frictionless.

If we imagine a channel set to `lorenz`, which produces 3 different values `x`, `y`, and `z`. On the channel itself we move between them with `morph` if I understand the present implementation. However, my suggestion is that for _other_ channels, the `out` waveform now shows not just channel 1 as an option - but all 3 of channel 1's `x`, `y`, and `z` outputs. In this way you can setup one channel to be the parent `lorenz` channel, and any number of others to monitor the different outputs (or combinations thereof).

The slight downside is that you can't directly monitor the `lorenz` channel, and I suppose you can't blend channel 8 with, say, channel 1 `z`, but I think it's a nice trade-off.

ostenning, 2024-07-09:
@cardinal-bin Thanks for the suggestion! An issue with this is that if channel 5 is no longer set to Lorenz, then what happens to channel 4? 
The mapping breaks in this circumstance, especially if something like waveform modulation was enabled.

cardinal-bin, 2024-07-09:
@ostenning Oh good point - I suppose the only way would be to have an `out` channel set to any of the 'sub channels' just jump to the parent channel if the waveform changes. I suppose any situation where linked channels are specific to a waveform (i.e. Lorenz) you might well have similar issues.

ostenning, 2024-07-09:
> @ostenning Oh good point - I suppose the only way would be to have an `out` channel set to any of the 'sub channels' just jump to the parent channel if the waveform changes. I suppose any situation [...]

Im considering removing the waveform modulation feature. To me it feels too experimental #78 and adds significant overhead. If I do then this feature could work, the user can still manually break the connection but providing the info bubble informs the user I think it would be OK