mnemonicdevices/md001 #78: Revise or remove waveform modulation

ostenning, 2024-07-09:
Waveform modulation changes the waveform based on a modulating parameter.

There has been some difficulty with this feature, its not implemented particularly well, and it adds a large complexity overhead preventing more advanced features from being added. 

Personally I don't find this feature useful, as it creates too unpredictable results. So I'm considering removing it altogether, or at the very least changing the way it works so that only on trigger events the waveform shape changes (rather than using LIN or EXP curves).

cardinal-bin, 2024-07-10:
Perhaps to avoid the issues with changing UI etc, a new waveform type could be added which uses its `morph` parameter to smoothly fade between a useful selection of the other waves. To maintain more of the existing functionality, a second `morph` parameter (like with envelopes) could be used to access some of the other waves normal morph functions.

ost-ing, 2024-10-24:
Waveform modulation was removed in 0.6.0. It would be nice to revise this functionality from the ground up to potentially reintroduce it.