mnemonicdevices/md001 #88: Recorder quantization on start

ost-ing, 2024-09-29:
It would be great though if there was an option for similarly quantising record to START on the next beat. And playback also to start on next beat. You could also have an option for start of next (4/4) bar, but then you’d need to be able to see if the bar was synced with your sound source – at least quantising to the beat option would really help.

This is a standard behaviour on a live looper pedal like infinity (which syncs to midi - it records at start of bar).

As it is, the first time the loop plays back after record it will be in time. But if you ever stop and restart then the slightest imprecision when either you started record or you start play again means playback is out of sync. I’ve found this the be a problem.

Please let me know if this is already possible and I just missed how! Else please consider for next update, it could be an item in record mod menu, to choose record/playback on beat 1/4 (or on next beat, otherwise free)

This would really bring the live looping functionality up to par for reliable ease of use!

cardinal-bin, 2025-01-22:
Partially related to this, it would be great to be able to set which beat the 0-phase position is considered. Using an external clock there's always a bit of instability at the start which can set things not where you want them.

twhiston, 2025-07-09:
+1 for start quantization. I was quite surprised that it didn’t already have this upon exploring the module.