mnemonicdevices/md001 #97: Quicker copy/paste

cardinal-bin, 2024-11-19:
**Is your feature request related to a problem? Please describe.**
When copying/pasting channels, there are a lot of required button presses. Open channel menu on channel to copy, select copy, copy, select channel to paste to, open channel menu, select paste, paste. When copying one channel several times, this is a laborious process!

**Describe the solution you'd like**
A quicker method to copy & paste channels. The ideal would be to allow the channel buttons to paste after a channel is copied - this would allow very quick copying of a single channel to all others for example. This could be in addition to existing functionality, with a separate option on the channel menu (e.g. "copy to..."). Alternatively a button combination could allow quick copy/paste from either the channel screen or performance screen. On the channel screen there aren't many suitable button combinations, hold encoder + tap channel button (to paste to that channel from the active one) could work but interferes with changing between fine/course parameter changes.

ost-ing, 2024-11-19:
Hi, you can also press the blue buttons while in the channel menu, to jump between channels without needing to close and reopen the channel menu. Have you tried this?

cardinal-bin, 2024-11-19:
Ah, amazing! That does make things much easier. This is missing (unless I missed it) from the docs however! :)

closed 2025-01-15 by ost-ing