mnemonicdevices/md001 #103: Adding values does not seems to respect AMP settings

cgestes, 2025-01-13:
**Describe the bug**
EX8 takes a CV that oscillates between -1V and -2V

CH7 and CH8 are configured on IN / EX8, AMP=-10, OFF=0

When the CH8 operand is set to ADD with source = CH7, the output of CH8 no longer oscillates, the displayed signal stays high, and the CV output is a flat line at 5V.

Note that it is working as expected with AMP=10V

Also, note that being able to multiply the signal by a constant (to amplify it) would be super useful :)

**To Reproduce**
Steps to reproduce the behavior.

**Expected behavior**
CH8 output to be -EX8*2 so oscillating between 2V and 4V

**Versioning (please complete the following information):**
 - 0.6.3

ost-ing, 2025-01-14:
Thanks for the report @cgestes I'll look into this

ost-ing, 2025-01-14:
I've reproduced the problem. It seems the problem here is that the interpreted value from EX8 does not take into consideration bipolar values.

I'll fix this for 0.7.0. Thanks again for reporting.